dvc94ch/pycircuit #3: Use Z3 for solving placement constraints and layout optimization

dvc94ch, 2017-07-14:
- [ ] Export placement constraints into an optimization problem in Z3's smt-lib format

kasbah, 2017-11-28:
Could you go into a bit more detail of how this could work? Maybe a link to a research paper? I can sort of imagine it but it would be interesting read if someone has already done work like this.

I played a bit (mainly just thought about) using Z3 for code synthesis when I was working on an [experimental footprint editor](https://github.com/monostable/footwork) but I didn't fully really realize that constraint programming could be used for layout tasks. 

I found [this paper](https://dl.acm.org/citation.cfm?id=2967072) (and Python implementations: [1](https://gist.github.com/nmz787/10c60e76941a8e5de624454666ea65b3) and [2](https://gist.github.com/nmz787/ae4a6121ce4aa42c2caed5b062b97c70)) where they use an SMT solver for escape routing which is a bit different but also interesting.

dvc94ch, 2017-11-28:
I looked was reading parts of [Placement and Design Planning for 3D Integrated Circuits](http://cadlab.cs.ucla.edu/news/dissertation_final.pdf). Most literature seems to be focused on IC placement, I don't remember finding any which proposed using an SMT solver. This issue is more of a "explore idea further issue".

MonoSAT looks interesting, thanks for the link. I'd like to revive this project sometime. It's not very useful at the moment, since the graphviz export for example becomes a mess real quick. I was working on some improvements to that a while ago but development stalled...

My current area of interest are low cost inkjet printers for rapid electronics prototyping.

kasbah, 2017-11-28:
It's an interesting topic that I'd like to explore more at some point. Thanks for the link.

Regarding the Graphviz output, I [added some to SkiDL](https://nbviewer.jupyter.org/gist/kasbah/cedfcb65327c7efe1565eb68fed1c5b0) but it's just the most basic thing that wouldn't help much for complex circuits. Though for SkiDL I think if you treat sub-circuits as nodes as well, and can choose a "zoom level" it could actually become useful for more complex circuits as well. 

Any links to read about your current projects?

dvc94ch, 2017-11-28:
Yes, that was the approach I was taking too.

Recently I've been playing around with rust on the hifive board: https://github.com/dvc94ch/riscv-rust-toolchain
Other than that I've not had too much time...

On the topic of inkjet printing I've mainly been reviewing literature, maybe find a project I could use as a bachelor thesis next year.

https://kitspace.org looks really nice, could be a nice source of real world placement/routing problems.

kasbah, 2017-11-28:
Ah cool, sounds like fun. I dabbled in Rust a bit and Hifive is really cool. Does it have decent support for peripherals: ADC, SPI etc?

Thanks! Kitspace is where I spend most of my free time. I am trying to figure out how to get more people to put their projects up. Not needing to use Git/Github will be big when I get it done but other than that I am not sure. I am hoping the BOM tools I am developing will prove valuable enough to attract people. 

Regarding real world placement/routing problems GitHub might be a better bet for now. I put this in a presentation recently which is approximate data from a basic survey I repeated twice:

> Open hardware is growing!
> Feb 2017:
> 
>   - ~3000 KiCad projects on GitHub
>   - ~7000 Eagle projects on GitHub
>   - ~9000 shared projects on OshPark
> 
> Oct 2017: 
> 
>   - ~5500 KiCad projects on GitHub
>   - ~12000 Eagle projects on GitHub
>   - ~13000 shared projects on OshPark
> 

By comparison we currently have 32 projects on Kitspace :)

dvc94ch, 2017-11-28:
The hifive board doesn't have an ADC (or I2C, DMA), Watchdog and SPI support is still missing.

I'd put my projects up, but still have very limited design experience... Getting user attention is always the hard part, I'm sure there are many awesome dead projects on the internet. Have you tried writing guest blog posts for OSH, hackaday, EEVBlog etc..?

kasbah, 2017-11-28:
I say, go for it! Even if it's just a really simple project. I wrote a post
for Octopart a while back but not for the ones you mentioned. It's a good
idea.

dvc94ch, 2017-11-30:
Your paper revived my interest in the subject. I've found some time to continue researching SAT and it's applications.

The new experiment would be:
1. read a kicad netlist
2. annotate any information required (see [0])
3. generate constraints in a language like minizinc [1]
4. compile to flatzinc and translate to dimacs cnf
5. dimacs cnf + netlist graph information -> dimacs gnf
6. solve with monosat
7. translate result to kicad pcb
8. view result, update constraints cycle
9. postprocess with pykicad (add copper pours etc.)

First step would be to play with minizinc and monosat, solving standard problems like n-queens, sudoku, etc. and then moving on to graph problems.

[0] [Optimization of electronics component placement design on PCB using self organizing genetic algorithm (SOGA)](https://link.springer.com/article/10.1007/s10845-010-0444-x)
[1] [minizinc](http://www.minizinc.org/)

dvc94ch, 2017-11-30:
Routing is expressed as a topological problem, which uses less nodes in a graph (n = number of pads) than a grid representation (n = x * y * z). It is then converted into an extended rubber band sketch and finally into a geometric routing.

[0] Introduction: SURF: Rubber-band Routing System for Multichip Modules
[1] Topological routing: RUBBER-BAND BASED TOPOLOGICAL ROUTER
[2] Spoke insertion: Fast and Incremental Routability Check of A Topological Routing Using a Cut-based Encoding
[3] Geometric routing: Geometric Transformations for a Rubber-band Sketch

dvc94ch, 2017-12-06:
Modeling the pin assignment problem in Z3 prototype:

```python
from z3 import *
import itertools

## Device
class Pin(object):
    counter = 0

    def __init__(self, *funs):
        self.id = Pin.counter
        Pin.counter += 1

        self.device = None
        self.funs = funs

        # Solution
        self.v_assign = Int('assign' + str(self.id))
        self.v_busassign = Int('busassign' + str(self.id))

        # Pin properties
        self.v_fun = Int('func' + str(self.id))
        self.v_bus = Int('bus' + str(self.id))

    def get_fun_by_name(self, name):
        for fun in self.funs:
            if fun.fun_name == name:
                return fun

    def pin_constraints(self):
        '''Pin constraints.'''

        _and = BoolVal(True)
        _or = BoolVal(False)
        for fun in self.funs:
            funid = self.device.funs_map[fun.fun_name]
            # Pin function can only be one of
            _or = Or(_or, self.v_fun == funid)
            if hasattr(fun, 'bus_name'):
                busid = self.device.buss_map[fun.bus_name]
                # If a pin has a function assigned it also has a bus assigned
                _and = And(_and, Implies(self.v_fun == funid,
                                         self.v_bus == busid))
                _and = And(_and, Implies(self.v_fun != funid,
                                         self.v_bus != busid))

        return And(_and, _or)

    def debug(self, model):
        return 'pin: %s assign: %s busassign: %s bus: %s func: %s\n' % \
            (self.id,
             str(model[self.v_assign]), str(model[self.v_busassign]),
             str(model[self.v_bus]), str(model[self.v_fun]))

    def eval(self, model):
        fun = model[self.v_fun].as_long()
        assign = model[self.v_assign].as_long()
        fun_name = self.device.funs[fun].fun_name
        return 'pin: %s func: %s assign: %s\n' % \
            (self.id, str(self.get_fun_by_name(fun_name)), str(assign))

class Device(object):
    def __init__(self, pins):
        self.pins = []
        self.buss = []
        self.buss_map = {}
        self.funs = []
        self.funs_map = {}

[... 136 lines cut ...]
for assign in assignments:
    for assign in assign:
        s.add(device.assignment_constraints(assign))

# Force a specific pin
# s.add(device.pins[0].v_assign == 3)

print(s.check())

model = s.model()

# print(device.debug(model))
print(device.eval(model))
```

```sh
time python3 pinassign.py
sat
pin: 0 func: GPIO assign: 2
pin: 1 func: UART0 UART_RX assign: 4
pin: 2 func: UART1 UART_TX assign: 3
pin: 3 func: GPIO assign: 1

python3 pinassign.py  0.46s user 0.02s system 99% cpu 0.488 total
```

closed 2017-12-13 by dvc94ch