dvc94ch/pycircuit #11: Setup and running examples

kasbah, 2017-12-13:
Hey, I just had my first go at trying to run this. I am assuming it is for python3 as I get `SyntaxError`s with python2. So I have done:

```bash
virtualenv venv3 -p/usr/bin/python3 && source venv3/bin/activate
pip install . 
```
 When I run the example from the Readme it gives:

## Devices
```
Traceback (most recent call last):
  File "examples/device.py", line 6, in <module>
    Pin('XTAL_XI', Fun('XTAL', 'XI')),
  File "/home/kaspar/projects/pycircuit/venv3/lib/python3.5/site-packages/pycircuit/device.py", line 35, in __init__
    self.bus = Bus.bus_by_type(bus_or_name)
  File "/home/kaspar/projects/pycircuit/venv3/lib/python3.5/site-packages/pycircuit/device.py", line 124, in bus_by_type
    raise IndexError('No Bus with type ' + type)
IndexError: No Bus with type XTAL
```

## Footprints

```
Traceback (most recent call last):
  File "examples/footprints.py", line 19, in <module>
    Map(17, 'GND'))
  File "/home/kaspar/projects/pycircuit/venv3/lib/python3.5/site-packages/pycircuit/footprint.py", line 36, in __init__
    self.device = Device.device_by_name(device)
  File "/home/kaspar/projects/pycircuit/venv3/lib/python3.5/site-packages/pycircuit/device.py", line 343, in device_by_name
    raise IndexError('No Device with name ' + name)
IndexError: No Device with name MCU
```

## Assign footprints

```
Traceback (most recent call last):
  File "assign.py", line 3, in <module>
    circuit = top()
NameError: name 'top' is not defined
```

And I haven run the remaining ones because they seem to require `circuit`.

dvc94ch, 2017-12-13:
The examples in the README are parts from `examples/small`. To run the examples do `cd examples/small` and run `make all`

kasbah, 2017-12-13:
Hmm, looks like the module isn't properly exported.

``` 
$ make all
make build
make[1]: Entering directory '/home/kaspar/projects/pycircuit/examples/small'
python3 small.py
Traceback (most recent call last):
  File "small.py", line 6, in <module>
    from pycircuit.formats import *
ImportError: No module named 'pycircuit.formats'
Makefile:4: recipe for target 'build' failed
make[1]: *** [build] Error 1
make[1]: Leaving directory '/home/kaspar/projects/pycircuit/examples/small'
Makefile:18: recipe for target 'all' failed
make: [all] Error 2 (ignored)
make place
make[1]: Entering directory '/home/kaspar/projects/pycircuit/examples/small'
rm -f small.out.pcpl
python3 ../../placer/place.py small.pcpl
Traceback (most recent call last):
  File "../../placer/place.py", line 1, in <module>
    from z3 import *
ImportError: No module named 'z3'
Makefile:7: recipe for target 'place' failed
make[1]: *** [place] Error 1
make[1]: Leaving directory '/home/kaspar/projects/pycircuit/examples/small'
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
```

If I copy the source directory over

```
$ cp -r ../../pycircuit/ .
$ make all
make build
make[1]: Entering directory '/home/kaspar/projects/pycircuit/examples/small'
python3 small.py
Traceback (most recent call last):
  File "small.py", line 154, in <module>
    pcb.from_pcpl('small.out.pcpl')
  File "/home/kaspar/projects/pycircuit/examples/small/pycircuit/formats/pcpl.py", line 28, in from_pcpl
    with open(filename) as f:
FileNotFoundError: [Errno 2] No such file or directory: 'small.out.pcpl'
Makefile:4: recipe for target 'build' failed
make[1]: *** [build] Error 1
make[1]: Leaving directory '/home/kaspar/projects/pycircuit/examples/small'
Makefile:18: recipe for target 'all' failed
make: [all] Error 2 (ignored)
make place
make[1]: Entering directory '/home/kaspar/projects/pycircuit/examples/small'
rm -f small.out.pcpl
python3 ../../placer/place.py small.pcpl
Traceback (most recent call last):
  File "../../placer/place.py", line 1, in <module>
    from z3 import *
ImportError: No module named 'z3'
Makefile:7: recipe for target 'place' failed
make[1]: *** [place] Error 1
make[1]: Leaving directory '/home/kaspar/projects/pycircuit/examples/small'
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
```

If I install z3:

```
$ pip install z3
$ make all
make build
make[1]: Entering directory '/home/kaspar/projects/pycircuit/examples/small'
python3 small.py
Traceback (most recent call last):
  File "small.py", line 154, in <module>
    pcb.from_pcpl('small.out.pcpl')
  File "/home/kaspar/projects/pycircuit/examples/small/pycircuit/formats/pcpl.py", line 28, in from_pcpl
    with open(filename) as f:
FileNotFoundError: [Errno 2] No such file or directory: 'small.out.pcpl'
Makefile:4: recipe for target 'build' failed
make[1]: *** [build] Error 1
make[1]: Leaving directory '/home/kaspar/projects/pycircuit/examples/small'
Makefile:18: recipe for target 'all' failed
make: [all] Error 2 (ignored)
make place
make[1]: Entering directory '/home/kaspar/projects/pycircuit/examples/small'
rm -f small.out.pcpl
python3 ../../placer/place.py small.pcpl
Traceback (most recent call last):
  File "../../placer/place.py", line 77, in <module>
    result = place(input)
  File "../../placer/place.py", line 11, in place
    box = Z3Box.from_dict(node)
  File "/home/kaspar/projects/pycircuit/placer/box.py", line 84, in from_dict
    return cls(int(d['id']), d['width'], d['height'])
  File "/home/kaspar/projects/pycircuit/placer/box.py", line 92, in __init__
    self.const_rx = int(math.ceil(width / 2))
NameError: name 'math' is not defined
Makefile:7: recipe for target 'place' failed
make[1]: *** [place] Error 1
make[1]: Leaving directory '/home/kaspar/projects/pycircuit/examples/small'
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
```

kasbah, 2017-12-13:
I am slowly getting there. Just compiling monosat. Will send a pull-request with fixes. 

It would be nice to have the examples in the README be runnable in ipython or whatever if this is installed. I think that's what people would expect. Can I change the README?

dvc94ch, 2017-12-13:
This is what I have in my `.zshrc` file:
`export PYTHONPATH=$PYTHONPATH:~/repos/pykicad:~/repos/pycircuit2:~/repos/monosat/src/monosat/api/python:~/repos/build/z3/python`

kasbah, 2017-12-13:
Ok, I managed to run it fully by installing the monosat python package and copying the compiled libmonosat.so into it ($PYTHONPATH would do it too). Looks like it can't solve the routing though, is that expected?

```
Solving in Monosat...
s UNSATISFIABLE
Makefile:11: recipe for target 'route' failed
make[1]: *** [route] Error 1
make[1]: Leaving directory '/home/kaspar/projects/pycircuit/examples/small'
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
```

dvc94ch, 2017-12-13:
Yes, as I mentioned, to solve the routing multilayer support is required since the packing is too dense.

> But I agree the documentation isn't very good, it's still all very unstable, and the example doesn't autoroute unless you disable this [line](https://github.com/dvc94ch/pycircuit/blob/master/pycircuit/formats/pcrt.py#L43) because there isn't multilayer support yet and the packing is to dense #7

closed 2017-12-26 by dvc94ch