dvc94ch/pycircuit #14: Use elkjs (or similar) for schematic-like drawings

kasbah, 2017-12-15:
[image]

I just came across [Netlistsvg](https://github.com/nturley/netlistsvg) which uses [Klayjs](https://github.com/OpenKieler/klayjs) for rendering Yosys netlists as schematics. Looks like [Elkjs](https://github.com/OpenKieler/elkjs) is supposed to replace Klayjs. It's a JS port of the [Eclipse Layout Kernel](https://www.eclipse.org/elk/downloads.html). I haven't found any Python implementations yet.

dvc94ch, 2017-12-15:
We don't need a python implementation. We can export json and modify netlistsvg to render it inside the viewer. We can probably add a export to svg or png feature inside the viewer. Or something along those lines. Thank you for the link!

kasbah, 2017-12-15:
Yes, looking over it, outputting the subset of the Yosys netlist that Netlistsvg supports and adding some SVG "skins" for analog components could be a good way to go for a proof of concept. What do you mean by "the viewer" though?

An interactive viewer that lets you "zoom" into sub-circuits would be really cool. Worth noting from the Netlistsvg readme:

> Yosysjs and klayjs are both capable of handling sub circuits and subgraphs. It might be interesting to see what kind of images I can produce from a hierarchal design.

dvc94ch, 2017-12-15:
> What do you mean by "the viewer" though

There is kind of a prototype in the viewer subfolder. You can start it with `node viewer/app.js net.svg pcb.svg` it's not very user friendly yet... pcb.svg doesn't have any styling built in it's in `viewer/css/pcb.svg.css`, the net highlighting is currently broken...

kasbah, 2017-12-15:
Ah, completely missed that. Very cool! The style for the Graphviz output makes more sense now.

kasbah, 2017-12-22:
I have started working on this as https://github.com/nturley/netlistsvg/pull/6/.

dvc94ch, 2018-01-02:
Started work on this, like the results so far. this depends on #4, since port directions need to be specified and checked for consistency. This also allows removing the component labels, since they are no longer needed, new `Inst('R', '10K')` and `SubInst(joule_thief())`.

dvc94ch, 2018-01-08:
Still pretty awful, using netlistsvg master. Correctly and consistently setting input and output information should make them look much better.

Power to signal or power to power nets should use vertical components, while signal to signal nets should use horizontal components. (analysis mostly implemented)

Port swapping and better skins should also make it look much better.

[image]
[image]
[image]

kasbah, 2018-01-09:
You should add these examples to netlistsvg and open issues for features you mentioned that haven't been opened yet.

nturley, 2018-01-09:
Hi, @dvc94ch . You raise some good points.

> Port swapping and better skins should also make it look much better.

Port swapping is definitely a problem. I'm still trying to figure out the rules for when to port swap. Better skins meaning more components?

>Power to signal or power to power nets should use vertical components, while signal to signal nets should use horizontal components

Right now I've been using a vertical flow for the analog skin, which does a bad job with horizontal ports so I've been just trying to use as many vertical components as possible, but presumably a typical circuit has more signal nets than power/gnd nets so maybe we should switch to a horizontal flow. Now that I have the lateral edge thing figured out, I might be able to make it look good. I'll try some experiments and see how it looks.

> The generic component should allow connecting nets at the top and bottom too

A simple fix would be to add some special info in the netlist for generic components, but I just tagged all the lateral ports in the skin, so maybe I should just start tagging all of the port directions in the netlist and use that to determine which ports are lateral.

Example 1:
----
[image]
Laterals branch fixed this problem. It just merged to master. Wires to horizontal ports were always messy until that merged.

---------

[image]
This needs a port swap. The resistor is connected from a southward port to a lateral port so we should be able to determine the optimal direction. The fact that the southward port is VCC should be an even bigger hint about the correct resistor orientation.

-------

[image]
Laterals branch does a better job with this one as well.

Example 2:
----------
[image]

Laterals branch fixed this

-----------
[image]

Turn off LONGEST_PATH. I think I turned it back off on master. I still think that NETWORK_SIMPLEX is better for analog.

Example 3:
------------
[image]
port swap, port swap, port swap.

So I think the issues for these layouts
* Port swap specific components (Especially when they are connected to VCC or GND) https://github.com/nturley/netlistsvg/issues/17
* See if we can get a better layout with a horizontal flow https://github.com/nturley/netlistsvg/issues/31
* Add more components in the skin (I don't think we need an issue for this because this will never be complete)
* Generics should have a way to get vertical ports https://github.com/nturley/netlistsvg/issues/30

dvc94ch, 2018-01-10:
Progress so far:
Things that can still be done from the pycircuit side are the following:
- [ ] custom logic for port swapping transformer - needs to add dots to symbol to show current flow (joule_thief)
- [ ] improve finding circuit paths on signal nets (sallen key without psu example can't determine that the resistor needs port swapping)
- [ ] improve finding circuit paths for circuits without external ports (sallen key with psu connected)

[image]
[image]
[image]
[image]
[image]
[image]

closed 2018-01-13 by dvc94ch