dvc94ch/pycircuit #32: Unable to install or use any example.

HTerminal, 2020-06-20:
Hello @dvc94ch 

I am trying to build this pycircuit the setup installs the pycircuit fine with this command.

`pip3 install pycircuit`  or `python3 setup.py install`

Am i doing it right can you please help me with how i can install and build and run any example. 
I have also tried the electron-lang project but unable to follow any documentation there. 
Please let me know.

Regards
Hitesh

dvc94ch, 2020-06-20:
Sorry both projects are unmaintained and will be for the foreseeable future. There's no money in eda, so I moved on to blockchain

HTerminal, 2020-06-20:
Ok @dvc94ch 
Thanks for the reply. 
Still any advice on how to get start with these projects.

dvc94ch, 2020-06-20:
Do you have an error message?

HTerminal, 2020-06-20:
Yes. Let me share that with you in a moment.

HTerminal, 2020-06-20:
So, @dvc94ch  here is my complete process i followed with error.
**Step 1 :**  I downloaded the github repo of pycircuit

**Step 2 :** I installed pycircuit using this command `python3 setup.py install`
which resulted this output

```
running install
running bdist_egg
running egg_info
writing pycircuit.egg-info/PKG-INFO
writing dependency_links to pycircuit.egg-info/dependency_links.txt
writing requirements to pycircuit.egg-info/requires.txt
writing top-level names to pycircuit.egg-info/top_level.txt
reading manifest file 'pycircuit.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'pycircuit.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build/lib
creating build/lib/placer
copying placer/__init__.py -> build/lib/placer
copying placer/bin.py -> build/lib/placer
copying placer/box.py -> build/lib/placer
copying placer/grid.py -> build/lib/placer
copying placer/place.py -> build/lib/placer
creating build/lib/pycircuit
copying pycircuit/__init__.py -> build/lib/pycircuit
copying pycircuit/build.py -> build/lib/pycircuit
copying pycircuit/circuit.py -> build/lib/pycircuit
copying pycircuit/compiler.py -> build/lib/pycircuit
copying pycircuit/component.py -> build/lib/pycircuit
copying pycircuit/device.py -> build/lib/pycircuit
copying pycircuit/diffev.py -> build/lib/pycircuit
copying pycircuit/layers.py -> build/lib/pycircuit
copying pycircuit/optimize.py -> build/lib/pycircuit
copying pycircuit/outline.py -> build/lib/pycircuit
copying pycircuit/package.py -> build/lib/pycircuit
copying pycircuit/pcb.py -> build/lib/pycircuit
copying pycircuit/pinassign.py -> build/lib/pycircuit
copying pycircuit/testbench.py -> build/lib/pycircuit
copying pycircuit/traces.py -> build/lib/pycircuit
creating build/lib/router
copying router/__init__.py -> build/lib/router
copying router/monosat.py -> build/lib/router
copying router/router.py -> build/lib/router
creating build/lib/pycircuit/formats
copying pycircuit/formats/__init__.py -> build/lib/pycircuit/formats
copying pycircuit/formats/json.py -> build/lib/pycircuit/formats
copying pycircuit/formats/kicad.py -> build/lib/pycircuit/formats
copying pycircuit/formats/spice.py -> build/lib/pycircuit/formats
copying pycircuit/formats/svg.py -> build/lib/pycircuit/formats
copying pycircuit/formats/yosys.py -> build/lib/pycircuit/formats
creating build/lib/pycircuit/library
copying pycircuit/library/__init__.py -> build/lib/pycircuit/library
copying pycircuit/library/components.py -> build/lib/pycircuit/library
copying pycircuit/library/design_rules.py -> build/lib/pycircuit/library
copying pycircuit/library/devices.py -> build/lib/pycircuit/library
copying pycircuit/library/outlines.py -> build/lib/pycircuit/library
copying pycircuit/library/packages.py -> build/lib/pycircuit/library
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/placer
copying build/lib/placer/__init__.py -> build/bdist.linux-x86_64/egg/placer
copying build/lib/placer/bin.py -> build/bdist.linux-x86_64/egg/placer
copying build/lib/placer/box.py -> build/bdist.linux-x86_64/egg/placer
copying build/lib/placer/grid.py -> build/bdist.linux-x86_64/egg/placer
copying build/lib/placer/place.py -> build/bdist.linux-x86_64/egg/placer
creating build/bdist.linux-x86_64/egg/pycircuit
copying build/lib/pycircuit/__init__.py -> build/bdist.linux-x86_64/egg/pycircuit
[... 112 lines cut ...]
Best match: pyparsing 2.4.6
Adding pyparsing 2.4.6 to easy-install.pth file

Using /usr/lib/python3/dist-packages
Finished processing dependencies for pycircuit==0.0.2
```

**Step 3 :** I cd into `pycircuit-master/examples/voltage_divider`

**Step 4: ** i run the example using this command `python3 voltage_divider.py`
which resulted in this output error
```
Traceback (most recent call last):
  File "voltage_divider.py", line 5, in <module>
    from placer.place import Placer
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.8/dist-packages/pycircuit-0.0.2-py3.8.egg/placer/__init__.py", line 1, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "/usr/local/lib/python3.8/dist-packages/pycircuit-0.0.2-py3.8.egg/placer/place.py", line 3, in <module>
ImportError: cannot import name 'json' from 'pycircuit.formats' (/home/eliot/.local/lib/python3.8/site-packages/pycircuit/formats/__init__.py)
```
  
Please if you have some time help me getting them start. 
Thank You 
Hitesh

HTerminal, 2020-06-21:
Hello @dvc94ch did you get any chance to look into it i also tried the methods of @kasbah. But they also did not worked for me.

kasbah, 2020-06-21:
What do you mean by "methods of @kasbah"? 

https://github.com/google/pcbdl might be worth a look for a more actively developed project for designing circuits using Python.

HTerminal, 2020-06-21:
Hello @kasbah i was actually mentioning this issue "https://github.com/dvc94ch/pycircuit/issues/11" opened by you. 
Also i found you have a lot of repository regarding PCB design using python. So i mentioned you too. Hope you don't mind.

Also i have tried to run examples of PCBDL . The examples of PCBDL are compiling and running but i am unable to find the outputs generated. by python.

Let me explain myself clearly. Actually i am working on this field from last few days. I also saw your video where you explained all the possible way like skidl , pycircuit and other.

I tired using skidl it worked fine. But unable to generate layouts. So kinda just searching for ways. Pycircuit looked promising but as @dvc94ch mentioned. He is not working on them.

So i am really looking for some guidance.

kasbah, 2020-06-21:
No problem @HTerminal. Glad you found the talk useful. I would open issues with PCBDL, the developer was very responsive on the issues I raised before when I wanted to try it out. I know people use skidl with success as well, so if want to use that but can't get it to work, open issues there. 

Otherwise, if you want to use pycircuit, looks like you would be a bit on your own trying to figure out how to get it to work.

HTerminal, 2020-06-21:
Hello @kasbah Thank you. I will open issues there.

Also we are a team working on a project related to this project. We are really looking forward to work with a person who has experience in this field. 
Can we have further conversation on email if you don't mind. I would like to share some brief about our work with you, we really want to move forward in this field.
I hope you don't mind @dvc94ch .  I know this seems not so professional talking here.

Really looking forward.